intel/rohd #12: Create additional simpler constructors to decrease verbosity of simple control blocks

mkorbel1, 2021-09-23:
The default implementation for `If`, `Iff`, etc. take lists of `Conditional`s as inputs, which can be excessively verbose in cases where there is only one `Conditional`.  Adding a special constructor (e.g. something like `Iff.s`) which takes a single `Conditional` could make simple code look cleaner.

quekyj, 2022-11-24:
```
If.s(condition, then, [orElse])
```

quekyj, 2022-11-24:
counter_test.dart
```
Sequential.multi([
      SimpleClockGenerator(10).clk,
      reset
    ], [
      If(reset, then: [
        val < 0
      ], orElse: [
        If(en, then: [val < nextVal])
      ])
    ]);
```

chykon, 2023-02-03:
Shouldn't this issue be closed after this (https://github.com/intel/rohd/pull/210) merge?

mkorbel1, 2023-02-03:
Yes, you're right, thank you @chykon!

closed 2023-02-03 by mkorbel1