intel/rohd #42: Support multiple triggers on flops

mkorbel1, 2021-11-05:
**Is your feature request related to a problem? Please describe.**
Support multiple triggers for flops, like in SystemVerilog:
```SystemVerilog
always_ff @(posedge clk or posedge reset)
```

**Describe the solution you'd like**
The `FF` and `FlipFlop` should have a syntax that accepts a list of signals instead of a single `clk` signal.

**Describe alternatives you've considered**
Using something like `clk | reset` is insufficient because there could be a rising edge of reset during a clock high period.

wswongat, 2021-12-04:
@mkorbel1 Is it possible to support negative edge trigger as well?
It seems like the rohd only support the positive edge single trigger with current version.
https://github.com/intel/rohd/blob/eb4fd660e9f4bb646c76f1bb7b900a445cc9ec79/lib/src/modules/conditional.dart#L222

mkorbel1, 2021-12-04:
@wswongat it wouldn't be too hard to add that, but I figured it wasn't really needed as a separate feature since you could always just use the posedge of an inverted signal, for example:
```dart
FF(clk, [...]); //posedge of clk
FF(~clk, [...]); // negedge of clk
```

Do you think that meets your needs or can you elaborate why you think a separate negedge trigger would be beneficial beyond this?

wswongat, 2021-12-04:
@mkorbel1 Your suggestion looks good. Looks like we need to find some way to support multiple triggers on flops

mkorbel1, 2021-12-06:
@wswongat #65 should address this issue and provide support for multiple triggers on flops

closed 2021-12-06 by mkorbel1