intel/rohd #93: An equality operator for `LogicValues` that supports "don't care"

mkorbel1, 2022-03-24:
## Motivation
There are cases where you want to compare two `LogicValues` values for some bits, but not all bits.  It would be nice to support an equality check that enables a comparison of two `LogicValues` where any `x` is treated as a "don't care if this bit is equal".  Some thought would need to be put into whether only one side of the comparison should consider `x` as don't care or both sides, or maybe that should be configurable.  This feature is similar to a `casez` but for values instead of hardware.

## Desired solution
Add a function for comparison with don't care to `LogicValues`.

## Alternatives considered  
Let users implement this functionality themselves as needed.

mkorbel1, 2022-03-28:
Comparison operator could also do things like compare a bus full of "x" to `LogicValue.x` with some sort of "fill" mechanism

dmetis, 2022-12-21:
I would like to take this one.

PR #294 (2023-03-07): First implementation for don't care scenarios
## Description & Motivation 

There are cases where you want to compare two LogicValues values for some bits, but not all bits. It would be nice to support an equality check that enables a comparison of two LogicValues where any x is treated as a "don't care if this bit is equal". Some thought would need to be put into whether only one side of the comparison should consider x as don't care or both sides, or maybe that should be configurable. This feature is similar to a casez but for values instead of hardware.

## Description of changes, and motiv [... cut]
closed 2023-03-07 by mkorbel1