intel/rohd #100: Create a utility that can convert SystemVerilog modules into ROHD

mkorbel1, 2022-03-29:
## Motivation
There is a lot of legacy SystemVerilog code.  ROHD offers a lot of benefits, not all of which are entirely realized purely through cosimulation.

## Desired solution
Something that can parse a SystemVerilog design and generate ROHD code from it.  If it can get to 100%, that would be ideal, because then it could be used in an automated flow.  Otherwise, if it can get most of the way there, it's at least something people can use to get started.

## Additional details
Here's some links to parsers that might be helpful:
https://github.com/alainmarcel/Surelog
https://github.com/google/verible

There's also CIRCT, which would be nice to leverage.

quekyj, 2023-03-14:
Some other references:
https://github.com/Nic30/hdlConvertor/tree/master
https://github.com/Nic30/hdlConvertorAst
https://github.com/MikePopoloski/slang

mkorbel1, 2026-09-29:
An interesting project that attempts to tackle this: https://github.com/maifeeulasad/sv2rohd