intel/rohd #104: Add checks to prevent the same `Conditional` from being used multiple times

mkorbel1, 2022-03-29:
## Motivation
If someone uses the same `Conditional` object multiple times in the same `_Always` or in different `_Always` blocks, it can cause unexpected behavior and simulation/synthesis mismatches.

## Desired solution
There should be an explicit check that `Conditional`s are not reused.  Perhaps a build-time check where each `Conditional` is assigned a parent `_Always`, and the `_Always` confirms there are no duplicates within itself?

quekyj, 2022-11-28:
@mkorbel1 I can pick this up as well.

quekyj, 2022-11-29:
Check at the construction time.

quekyj, 2022-11-30:
Conditional Object creation:
```
final condOne = If(a, then: [q < d]);
```

quekyj, 2022-12-01:
Write a test 

```
final condone = If(a, then: [q < d]);
```

1. Put this condone in 2 conditionals block
2. Add a test.

PR #227 (2023-01-04): [Issue-104] checks to prevent same conditional used multiple times

## Description & Motivation

If someone uses the same Conditional object multiple times in the same _Always or in different _Always blocks, it can cause unexpected behavior and simulation/synthesis mismatches.

## Related Issue(s)

Fix #104 

## Testing

Make sure an exception is thrown when conditionals are reuse.

## Backwards-compatibility

> Is this a breaking change that will not be backwards-compatible?  If yes, how so?

no

## Documentation

> Does the change require any updates to documentation?  If so, where?  Are the [... cut]
closed 2023-01-04 by mkorbel1