intel/rohd #111: Add extensive testing for zero-width `Logic`s and `LogicValue`s

mkorbel1, 2022-03-30:
## Motivation
There have been multiple bugs found related to zero-width `Logic`s and `LogicValue`s.

## Desired solution
Add a bunch of tests covering behavior of things when the are zero-width, both functionally and for synthesis.

mkorbel1, 2022-03-30:
See #57

mkorbel1, 2024-01-04:
Some things to consider:
- Unary operations (in gates and on `LogicValue`s) need to probably throw an error when called on a 0-width signal

dmetis, 2024-01-09:
I can take this

soneryaldiz, 2024-09-25:
It would also be convenient to have `isEmpty`  for `Logic` and `LogicValue`.

mkorbel1, 2024-09-25:
> It would also be convenient to have isEmpty for Logic and LogicValue.

You mean just as automation for `.width == 0`?

soneryaldiz, 2024-09-25:
Yes.

mkorbel1, 2024-10-15:
Additionally, swizzling of no signals (or only zero-width signals) should throw an exception.

closed 2026-11-03 by desmonddak