intel/rohd #114: Add a test against multiple sequential drivers on the same signal

mkorbel1, 2022-03-30:
## Motivation
It doesn't make sense for a signal to be driven by two drivers in a `Sequential` block.  The code already prevents this, but testing is incomplete.

## Desired solution
Add a test ensuring that an exception is thrown when there are multiple drivers for a flop.

quekyj, 2022-10-24:
Hi Max, I would gladly take over this issue next.

quekyj, 2022-10-31:
## Things to Note
``conditional_test.dart``: line 142, in the Sequential
``conditional.dart``: line 373, try to add something to the ``Set`` and expect to throw Exception

Basically, we want to make sure the ``conditional.dart`` not allow re-assignation of same variable, need to throw error 

## Non-blocking Assignment
```
throw Exception('Sequential drove the same signal(s) multiple times:'
            ' $redrivenSignals.');
```

quekyj, 2022-10-31:
conditional.dart
The ``element.execute()`` is in data structure of Set which already prevent duplication of driver. Not sure how to create a test for code below to throw the exception. 

```
else if (anyClkPosedge) {
      final allDrivenSignals = <Logic>[];
      for (final element in conditionals) {
        allDrivenSignals.addAll(element.execute());
      }
      if (allDrivenSignals.length != allDrivenSignals.toSet().length) {
        final alreadySet = <Logic>{};
        final redrivenSignals = <Logic>{};
        for (final signal in allDrivenSignals) {
          if (alreadySet.contains(signal)) {
            redrivenSignals.add(signal);
          }
          alreadySet.add(signal);
        }
        throw Exception('Sequential drove the same signal(s) multiple times:'
            ' $redrivenSignals.');
      }
```

My suggestion of the alteration is added the validation code to the line 528, as its would check for the errors during the execute of the conditionals under the class ConditionalAssign:

```
@override
  Set<Logic> execute() {
    // Add exception validation and check
    ///////////////////////////////////////////

    receiverOutput(receiver).put(driverValue(driver));
    return {receiver};
  }
```

PR #197 (2022-12-15): [Issue-114] Add test to sequential driver on same signal to return Exception

## Description & Motivation

It doesn't make sense for a signal to be driven by two drivers in a Sequential block. The code already prevents this, but testing is incomplete.

Also fixed bug on the previous exception won't run.

## Related Issue(s)

Fix #114 

## Testing
``dart test`` complete successfully. New test added to the ``conditionals_test.dart``.

```
test(
      'should return SignalRedrivenException when there are multiple drivers '
      'for a flop.', () async {
    final mod = SignalRedrivenSequentialMo [... cut]
closed 2022-12-15 by mkorbel1