intel/rohd #116: Better support for `Logic` based indexing of `List<Logic>`

mkorbel1, 2022-04-07:
## Motivation
In SystemVerilog, it's easy to index an entry in an array of signals:
```SystemVerilog
logic [8:0] a [4:0];
logic b [4:0];
logic[8:0] c;

assign c = a[b];
```

ROHD doesn't have a synthesizable array type, instead encouraging usage of `List<Logic>`.  However, this makes it much more verbose to access an index of an array based on another `Logic`.

## Desired solution
Some sort of function or operation which reduces the effort required to do this sort of operation.

## Alternatives considered  
It is possible by looping through possible values manually and generating comparisons and `or`ing them together, but that's more verbose.

mkorbel1, 2022-08-19:
Maybe a function on `Logic` called `select` or something that operates on an indexable `Iterable<Logic>` that automatically generates the hardware to select the right one?

RPG-coder-intc, 2023-02-08:
Hey Max, I will start working on this one :)

mkorbel1, 2023-02-17:
An example of an equivalent API in ROHD could be:
```dart
List<Logic> a; // length of list is 5, each element width is 9 bits
Logic b; // width is 3 bits
Logic c; // width is 9 bits

// two potential options:
c <= a.selectIndex(b); // order matches closer to array indexing
c <= b.selectFrom(a);
```

RPG-coder-intc, 2023-04-06:
Working on this issue...

mkorbel1, 2023-12-14:
Fixed in #438

closed 2023-12-14 by mkorbel1