intel/rohd #128: Make `LogicValue` `toInt`, `toBigInt`, etc. return `null` when signal is invalid instead of an Exception

mkorbel1, 2022-06-06:
## Motivation
Dealing with null is easier with Dart's null safety than an exception.

## Desired solution
Functions that might not have a valid int/BigInt representation due to invalid signal values (x/z) should return `int?` and `BigInt?` instead of throwing an exception when it is invalid.

## Alternatives considered  
Leave as it is with exceptions thrown when invalid.

mkorbel1, 2022-10-21:
It's worth considering which is better a little further.  It might be good to look at other Dart conventions for `toX` types of functions.

mkorbel1, 2023-01-31:
Instead of changing the behavior of the existing API, what if we add "try" versions like these:
https://api.flutter.dev/flutter/dart-core/int/parse.html
https://api.flutter.dev/flutter/dart-core/int/tryParse.html

For example, `LogicValue.tryToInt()`

priya-t12, 2023-01-31:
I will work on this.