intel/rohd #138: Custom definition names of `Module`s are not sanitized

mkorbel1, 2022-07-26:
## Describe the bug
It is possible to specify a `definitionName` that is not sanitary (as sanitized by `Sanitizer`), which means you can generate invalid SystemVerilog.

## To Reproduce
Define a module with an unsanitary `definitionName` passed to `Module`'s constructor.

## Expected behavior
ROHD sanitizes the definition name so that it generates valid SystemVerilog.

## Actual behavior
The unsanitary definition name is included directly in the generated SystemVerilog.

**Additional context**  
Sanitization may as well occur directly within the constructor.

quekyj, 2022-10-18:
Hi @mkorbel1 , I will have a look into this.

quekyj, 2022-11-03:
Issue resolved.