intel/rohd #141: Add support for `++` and `--` types of operations in conditional logic

mkorbel1, 2022-07-27:
## Motivation
There are cases in procedural execution in `_Always` blocks where `++` and `--` are useful.  Dart does not allow overloading these operators, however.

## Desired solution
Some sort of function on `Logic` which acts like `++` and `--`, both postfix and prefix (i.e. `++x` vs. `x++`).  Some thought has to go into the API and function names to make this intuitive.

To effectively use these requires some careful thought.  A prefix is simple, i.e. `++x` is the same thing as `x + 1`.  A postfix requires that the original signal is returned, but then a modified conditional operation is performed afterwards.  This means that it needs to return some special type of object that other `Conditional`s can handle, but that also might need to play nicely with other `Logic` manipulation within conditional procedural logic.  There's a number of combinations to consider, and this could involve some modifications to the way things like `ConditionalAssignment`s work in general.

## Alternatives considered  
Just omit the feature.

## Additional details
There are real designs where users had wished for a feature like this.

mkorbel1, 2022-08-19:
Adding something like += and -= would also be good!

RPG-coder-intc, 2022-10-11:
Hey Max, I would like to give a try on this issue! :)

mkorbel1, 2022-10-11:
@RPG-coder-intc great!  Let me know once you have a strategy mapped out if you'd like to review the approach.  I expect there could be some tricky surprises in implementing this.

RPG-coder-intc, 2022-10-24:
A function will be made for postIncrement and preIncrement taking the Logic object value as a reference. The result computation will be worked on this reference object taken as a parameter.

RPG-coder-intc, 2022-10-24:
Similar pattern will be followed on `+=` (`addAssign(&ref)`) , -= (`diffAssign(&ref)`) , `*=` (`mulAssign(&ref)`), `/=` (`divAssign(&ref)`).

RPG-coder-intc, 2022-10-24:
`logicVar++` ===  `postIncr(&ref)`
`++logicVar` ===  `preIncr(&ref)`

chykon, 2022-10-24:
Perhaps it's better to use `subAssign` instead of `diffAssign`?

RPG-coder-intc, 2022-10-24:
> Perhaps it's better to use `subAssign` instead of `diffAssign`?

It can be done. What's your say @mkorbel1

mkorbel1, 2022-10-24:
Why not make `postIncr` and `preIncr` support another input (which defaults to 1) so that the same function can support `++` and `+=`?  Then the natural extension for subtraction would be `postDecr` and `preDecr`.

RPG-coder-intc, 2022-10-24:
> Why not make `postIncr` and `preIncr` support another input (which defaults to 1) so that the same function can support `++` and `+=`? Then the natural extension for subtraction would be `postDecr` and `preDecr`.

The would be a great addition.

closed 2023-01-31 by mkorbel1