intel/rohd #146: Support tagging of generated SystemVerilog `begin` and `end`

mkorbel1, 2022-08-11:
## Motivation
Some users have extra preferences about the readability of generated SystemVerilog for their applications.  It would be nice to use the tagging mechanism for if, case, etc. in SV to help along those lines.

## Desired solution
Every ‘begin’ should have the ability to add a name for it.

```Verilog
if (a) begin : if_name
   .
   .
end else begin : else_name
   .
   .
end // if_name
```

This should be simple to add to things like `Iff`, `ElseIf`, `Else`, as well as `If` and `Case`.

mkorbel1, 2022-12-22:
Some lint tool checks need this feature, so there is some urgency to get this implemented to make life easier for users producing SystemVerilog that passes through those tools.

lucasphillips, 2025-03-05:
As part of the 2025 OSS Hackathon, ROHD_Team1 is working on this issue.

mkorbel1, 2025-03-05:
Awesome, assigned!