intel/rohd #157: Make `signExtend` generate prettier SystemVerilog

mkorbel1, 2022-09-01:
## Motivation
Right now the `signExtend` genreates a mux, which is especially ugly when there's only 1 bit.  SystemVerilog has some nice syntax for repeating a signal value (replication operator) which could be nicer in the generated code.

## Desired solution
For example:
```SystemVerilog
logic x;

// instead of this
x[0] ? 8'hff : 8'h0

// do this
{8{x}}
```

There's special cases to consider for 1-bit values vs. >1-bit values, since you don't need to do `[0]` on a 1-bit signal to get the upper-most bit.

akshay-wankhede, 2022-10-04:
@mkorbel1 can you please assign this to me?

akshay-wankhede, 2022-11-21:
PR https://github.com/intel/rohd/pull/203 raised with the fix.

closed 2022-11-22 by mkorbel1