intel/rohd #161: A double "tick" is required to generate waveform

chykon, 2022-09-20:
## Describe the bug

A double `tick` is required to create a new timestamp.

## To Reproduce

Use code:

```dart
import 'package:rohd/rohd.dart';

class ExampleModule extends Module {
  ExampleModule(Logic inputA) {
    inputA = addInput('input_a', inputA);
    final out = addOutput('out');

    Combinational([
      out < inputA,
    ]);
  }

  Logic get out => output('out');
}

Future<void> main() async {
  Logic inputA = Logic();
  final exampleModule = ExampleModule(inputA);
  await exampleModule.build();
  WaveDumper(exampleModule);
  SimpleClockGenerator(2);
  for (var i = 0; i < 10; ++i) {
    inputA.inject(i % 2);
    await Simulator.tick();
    //await Simulator.tick();
    print(exampleModule.out.value);
  }
}
```

Open `waves.vcd` in viewer.

## Expected behavior

You can watch the history of signal changes.

## Actual behavior

There is no signal history.

## Additional details

```text
Dart SDK version: 2.18.0 (stable) (Fri Aug 26 10:22:54 2022 +0000) on "linux_x64"
```

```yaml
name: package

environment:
  sdk: '>=2.14.0 <3.0.0'

dependencies:
  rohd: ^0.3.2
```

## For @mkorbel1

I like how actively you participate in resolving my issues, but I do not want to be intrusive. In the course of using ROHD, I noticed some roughness, and also had ideas for improving some aspects. I will try to gradually share them with you.

As a result, I understand the limited resources of the ROHD project, so please process my questions and suggestions at your own pace. Personally, I have not encountered a single problem that would block further work. And yes, ROHD is just great.

mkorbel1, 2022-09-20:
@chykon thank you for the note!  I'm actually out of town at the moment so my time is unusually restricted.  We really appreciate the feedback and welcome suggestions and ideas!

I'll take a closer look at this one as soon as I have a moment.

mkorbel1, 2023-03-27:
The public `tick` function is currently a little confusing since it could require multiple ticks to execute one timestamp.

A more intuitive way to handle time progressing is to `await` clock edges after kicking off the `Simulator`.

However, it would be also nice to add some sort of function that has the behavior of simulating up until some event or timestamp, and then pausing.  I think this should be already achievable, but not in a convenient way, so this is effectively adding some automation to `Simulator` to make it easier.  For example, adding APIs like `Simulator.runUntil` which accepts a `Future` and/or a `Simulator.time` to stop at.

I suspect the waveform generation behavior described in this ticket is a symptom of the unintuitive `tick` behavior, but need to dig in a little more.

closed 2026-11-03 by desmonddak