intel/rohd #232: Add a mechanism for generating random `LogicValue`s

mkorbel1, 2022-12-20:
### Motivation

`LogicValue` is a useful type since it supports arbitrary width in an efficient representation.  For testing purposes, it might be desirable to generate a random value on a bus of arbitrary width.  The built-in Dart randomization only randomizes up to 2^32, which means users have to invent their own little code snippets to generate random numbers bigger than 32 bits.

### Desired solution

A way to generate large random numbers, especially for `LogicValue`.

One nice option might be to add an `extension` on `Random`.

It might be nice to support both valid values or a mix of valid and invalid values.

It might be nice to support random `BigInt`s as well.

quekyj, 2023-05-03:
```dart
Random().nextLogicValue(10);
```

quekyj, 2023-05-03:
arguments:
- width, maximum value, invalid bits (x and z) - true/false (bool)

PR #362 (2023-06-07): 232 add a mechanism for generating random logic values

## Description & Motivation

LogicValue is a useful type since it supports arbitrary width in an efficient representation. For testing purposes, it might be desirable to generate a random value on a bus of arbitrary width. The built-in Dart randomization only randomizes up to 2^32, which means users have to invent their own little code snippets to generate random numbers bigger than 32 bits.

## Related Issue(s)

Fix #232 

## Testing

Add unit test to test random generation of the `LogicValue`.

## Backwards-compatibility

> Is this a br [... cut]
closed 2023-06-07 by mkorbel1