intel/rohd #246: `SynthBuilder` should check modules are already built

mkorbel1, 2023-01-09:
### Describe the bug

It's possible to generate outputs using a `Synthesizer` and `SynthBuilder` rather than through `Module.generateSynth`.  However, the `SynthBuilder` does not check that `Module.build` is called, only `generateSynth` does.  This can produce unexpected outputs (e.g. SystemVerilog with no submodules) without any helpful error message indicating why.

### To Reproduce

Use `SynthBuilder` to generate output verilog without calling `build` on the `Module`.

### Expected behavior

`ModuleNotBuiltException` is thrown

### Actual behavior

Silent incorrect results

quekyj, 2023-03-27:
@mkorbel1 Let me look into this.

mkorbel1, 2023-03-27:
Sure, thanks!

PR #340 (2023-04-20): [Issue 246] SynthBuilder should check for Module.build()

## Description & Motivation

It's possible to generate outputs using a Synthesizer and SynthBuilder rather than through Module.generateSynth. However, the SynthBuilder does not check that Module.build is called, only generateSynth does. This can produce unexpected outputs (e.g. SystemVerilog with no submodules) without any helpful error message indicating why.

## Related Issue(s)

Fix #246 

## Testing

Tested with module not built and catch exception. 

## Backwards-compatibility

> Is this a breaking change that will not be backwards [... cut]
closed 2023-04-20 by mkorbel1