intel/rohd #270: Add automation for code coverage in PRs and pushes

mkorbel1, 2023-02-13:
### Motivation

Many repos have code coverage metrics provided in two places automatically:
- On the README as a badge, indicating coverage for the repo by the test suite.
- In PRs as an indication of how much coverage there is including the changes of the PR.
  - This can also sometimes include a change in coverage relative to the mainline.

Adding automation around these can have some nice benefits, for example:
- Demonstrate to users how well tested the repository is
- Identify testing gaps in new PRs more easily

It would be nice if ROHD (and related packages) had some automation around this.

### Desired solution

An automated coverage run that provides information about coverage in PRs and on the mainline.  It would be nice to generate the whole HTML report in a way that is easily inspectable.

### Alternatives considered

Some tools (even free ones) exist to do a lot of this, but using another software product integrated into the repository requires jumping through some additional hurdles.

Ideally, we are able to implement this without dependence on a 3rd party service.

chykon, 2023-02-21:
Can I take it? Are there any deadlines?

mkorbel1, 2023-02-21:
> Can I take it?

Sure, it's all yours, thanks!

> Are there any deadlines?

Nope, no firm deadline :)

closed 2026-11-03 by desmonddak