intel/rohd #295: Arithmetic shift-right generated SV can be ineffective depending on nearby logic

mkorbel1, 2023-03-02:
### Describe the bug

If an unsigned operation is applied to the result of an arithmetic shift-right in the generated SystemVerilog, it may perform as a logical (unsigned) right-shift instead.

For example, if generated SystemVerilog looks like this:
```SystemVerilog
  assign x = {32{c}} & $signed($signed(a) >>> b) ;
```
This actually performs an _unsigned_ shift right.

### To Reproduce

Perform an unsigned bitwise operation on the result of an arithmetic shift-right.

### Expected behavior

Generated verilog for arithmetic shift right is consistent always.

### Actual behavior

Depending on surrounding logic, a logical shift-right may occur instead.

### Additional: Context

Present in v0.4.2

saw235, 2023-03-04:
Can you try this one? Note the curly braces instead of the parenthesis. 
```verilog
 assign x = {32{c}} & {$signed(a) >>> b} ;
```
The reason for this is because the curly braces makes it self-determined. I suspect that the & operator causes it to think that the result type should be unsigned and thus the arithmetic shift makes a zero extend.

mkorbel1, 2023-03-06:
> Can you try this one? Note the curly braces instead of the parenthesis.
>
> ```verilog
>  assign x = {32{c}} & {$signed(a) >>> b} ;
> ```
>
> The reason for this is because the curly braces makes it [...]

Good call @saw235, I like this better.  Tests pass, updated PR with that adjusted.  Look good to you?
https://github.com/intel/rohd/pull/296

PR #296 (2023-03-08): Fix #295, signed arithmetic shift right

## Description & Motivation

Shift-right arithmetic (SRA) is broken in generated SystemVerilog if an unsigned operation is applied to the result (see #295)

This fix maintains that the SystemVerilog still uses `>>>` so that it is similar to the original implementation but adds an additional set of `{}` around the result to guarantee safety.

Thank you to @saw235 for the implementation suggestion in https://github.com/intel/rohd/issues/295#issuecomment-1454651572

## Related Issue(s)

Fix #295

## Testing

Added a new test that failed before the fix.

## [... cut]
commit 59a5af36 (2023-03-08): Fix #295, signed arithmetic shift right (#296)
closed 2023-03-08 by mkorbel1