intel/rohd #305: Don't use the GitHub Wiki

chykon, 2023-03-09:
### Motivation

The [ROHD Forum](https://github.com/intel/rohd/wiki/ROHD-Forum) currently uses the GitHub Wiki, which has a few downsides:

* [not indexed by search engines](https://docs.github.com/en/communities/documenting-your-project-with-wikis/about-wikis#:~:text=Note%3A%20Search%20engines%20will%20not%20index%20the%20contents%20of%20wikis.%20To%20have%20your%20content%20indexed%20by%20search%20engines%2C%20you%20can%20use%20GitHub%20Pages%20in%20a%20public%20repository.)
* limited collaborative editing (not so easy to track changes and not able to control formatting of Markdown files using GitHub Actions)
* you can't get Wiki update notifications when watching a repository
* limited ability to manage editing permissions: either for collaborators or for anyone with a GitHub account (may lead to vandalism) ([info](https://docs.github.com/en/communities/documenting-your-project-with-wikis/changing-access-permissions-for-wikis))

### Desired solution

Ideally, a separate repository could be used for `Forum` ([example](https://github.com/WebAssembly/meetings)), but you can do without it. Here are the suggested options:

* Move data related to `Forum` to eg `doc/forum`.
* Create a new GitHub Discussions thread that will include `Forum` related posts.
* Any of the options above, but applicable to the repository that will be used for the ROHD website. As I understand it, `Forum` is used to discuss the entire ROHD ecosystem, so it is proposed to place it on "neutral territory".

The current Wiki can be [hidden](https://docs.github.com/en/communities/documenting-your-project-with-wikis/disabling-wikis) so that neither users nor developers are exposed to outdated information. You will also need to update all links.

---

In addition to the `ROHD Forum`, the Wiki has information added by @quekyj. As far as I understand, this data also follows along with changes related to the site and tutorials, so it does not need to be transferred anywhere, and the Wiki can be safely hidden. Correct me if I'm wrong.

mkorbel1, 2023-03-09:
Thanks for the thought and insight!  You raise some good points.

What do you think of making the new repo that we were going to use for the website a more broad repo that could include the website, meeting minutes, and other miscellaneous documentation stuff in the future not directly appropriate for the main repository?  I don't know what it would be called though, and maybe that's just a lazy solution?  Hosting it on the website doesn't seem too inappropriate?

I'm a bit reluctant to spam the main repository's commit history with minutes and agendas.

The discussions idea is possibly good, but I'm unsure how to keep it well organized/categorized in there.  Do you mean just one discussion with a new comment added each meeting?

chykon, 2023-03-09:
> What do you think of making the new repo that we were going to use for the website a more broad repo that could include the website, meeting minutes, and other miscellaneous documentation stuff in the future not directly appropriate for the main repository?

This is a good idea. The new repository can be seen as a mono-repository for "non-main" sub-projects in the ROHD umbrella project.

> I don't know what it would be called though, and maybe that's just a lazy solution? Hosting it on the website doesn't seem too inappropriate?

How to call - it is necessary to think. We need something abstract. My suggestions:

* `rohd-mono`
* `monorohd`
* `rohd-misc`

I think this is a good way to avoid the bureaucracy (which you pointed out) associated with creating a new repository. A mono-repository may be more difficult to maintain, but maintaining many small repositories can also be difficult.

> The discussions idea is possibly good, but I'm unsure how to keep it well organized/categorized in there. Do you mean just one discussion with a new comment added each meeting?

Yes. I think it might look like this:

1. Create a new discussion `ROHD Forum` (zero message describes what it is, how to join)
2. A message is created that indicates the date and time of the event, topics for discussion
3. We are waiting for the end of the meeting
4. We supplement the message with a brief summary of the meeting
5. GOTO 2

If the idea of a mono-repository is accepted, then it can be placed there.

mkorbel1, 2023-03-23:
I think using submodules that point to ROHD, ROHD-VF, ROHD-Cosim, and any others that pop up could be a reasonable way to handle references to content in each of the repos.  However, that would mean each time we commit or release a version, we'd also need to go manually update the website.

An alternative could be some sort of cronjob that regenerates the static page based on latest `main` contents on each of the other projects.

Once the new repo is created, I guess we can experiment with different ways to manage this.

quekyj, 2023-03-24:
I think I will remove all the information inside Wiki once the bootcamp is complete. Btw, I am thinking that adding a FAQ section will be helpful as well.

mkorbel1, 2023-03-27:
The repo has been created here: https://github.com/intel/rohd-website

mkorbel1, 2023-08-02:
The contents on the wiki, including the forum, have fully migrated to the ROHD website!
https://intel.github.io/rohd-website/

closed 2023-08-02 by mkorbel1