intel/rohd #313: Specifying the type of logic `StateMachine`

chykon, 2023-03-14:
### Motivation

It might be worth adding a clarification to the documentation that `actions` in `State` (state `StateMachine`) contains combinational, not sequential logic. Personally, I thought that sequential logic is used for state actions, since the `StateMachine` interface contains `clk`, `reset` inputs.

### Desired solution

Add to the documentation a clarification that `actions` in `State` (state `StateMachine`) contains combinational, not sequential logic.

mkorbel1, 2023-03-14:
Good point, thanks for the suggestion!

mkorbel1, 2024-09-17:
This came up again, and upon discussion it seems like it might make sense to add the ability to provide either/both on the FSM.  For example, we could add something like `updates` as an additional `List<Conditional>` argument which updates sequentially instead of driving combinationally.

soneryaldiz, 2024-09-18:
`actions` and `updates` do not appear very intuitive yet I don't have a better suggestion yet. Is it possible to keep actions as-is and determine combinational versus sequential updates based on the operator such as = vs <= (motivated by blocking vs non-blocking assignments).

soneryaldiz, 2024-09-18:
No, let's allow combinational and/or sequential actions.

closed 2026-11-03 by desmonddak