intel/rohd #334: Add an enable signal to `FlipFlop`

mkorbel1, 2023-03-27:
### Motivation

Along the same lines as https://github.com/intel/rohd/issues/69

It would be nice to have an enable signal optionally provided as part of the built-in `FlipFlop`.

### Desired solution

Add an optional enable signal to `FlipFlop`.

gantanikhilraj, 2023-05-01:
Hello @mkorbel1 , could you brief me the issue a bit more?

mkorbel1, 2023-05-01:
Sure!  There's a `Module` called `FlipFlop` in `lib/src/modules/conditional.dart`.  The current constructor looks like this:
```dart
FlipFlop(Logic clk, Logic d, ...)
```

This represents a simple flip-flop sequential element which samples `d` on each rising edge and drives the last cycle's input on `q`.  An `enable` on a flip-flop is a way to control whether it should sample on the rising edge or continue to hold the previous value sampled when `enable` *was* high.

This enhancement is to add an optional `enable` signal (probably a nullable, optional named argument to the constructor) which then generates an enableable flip-flop.

Let me know if that helped clarify and if you'd like to take up the issue!

Sanchit-kumar, 2023-05-07:
I can take up this issue.

closed 2023-05-26 by mkorbel1