intel/rohd #377: Write part-assign automation for `LogicArray`s

mkorbel1, 2023-06-05:
### Motivation

Currently, to assign a group of elements in a `LogicArray`, you need to operate on each of `elements` you're interested in individually.  This is tiresome.

### Desired solution

Some automation that allows assigning a range or ranges of elements in a `LogicArray` all at once.

### Additional details

There's some discussion to be had on API design.

RPG-coder-intc, 2023-10-26:
Hey Max, I would like to take this one on

RPG-coder-intc, 2023-12-14:
reference: panda 

Case:
`data[i:j] = Const(defaultValue)`
`data[i:j] = [Const(value1), Const(value2)]` or LogicArray() of length (j-i+1)

mkorbel1, 2024-08-02:
Closed in #456

closed 2024-08-02 by mkorbel1