intel/rohd #384: Wave dumping with `LogicArray`s and `LogicStructure`s

mkorbel1, 2023-06-16:
### Motivation

#375 introduced `LogicArray`s and `LogicStructure`s, but did not test or consider the `WaveDumper`.  There are likely some modifications necessary, and definitely some tests required.

### Desired solution

Write tests for the `WaveDumper` considering different scenarios of `LogicArray` and `LogicStructure` and implement necessary upgrades to make everything work right.

mjayasim9, 2023-10-10:
Will work on this!

mjayasim9, 2023-11-07:
The example below shows the VCD file generated for a test with `LogicArray`

[image]
 

[image]

VCD currently does not support `LogicArray` representation (https://github.com/verilator/verilator/issues/3167) and therefore, we can close this issue for now. An alternate solution would be to split the bus by arrays, however, that would create numerous waves and it would be hard to view them unless there is a search option in the Wave Viewer.

mkorbel1, 2023-11-07:
Thanks for looking into all this @mjayasim9!  Yes, we can close this for now since VCD doesn't seem to have any nice support for arrays anyways.

closed 2023-11-07 by mkorbel1