intel/rohd #394: Generate RTL for all classes even when two classes have identical contents

sshankar4, 2023-07-24:
### Describe the bug

Two classes with same contents:

class my_mod extends Module {
  my_mod(Logic clk, {String name = "my_mod"})
      : super(
            name: name, definitionName: "my_mod", reserveDefinitionName: true) {
    addInput('clk', clk);
  }
}

class my_mod_copy extends Module {
  my_mod_copy(Logic clk, {String name = "my_mod_copy"})
      : super(
            name: name,
            definitionName: "my_mod_copy",
            reserveDefinitionName: true) {
    addInput('clk', clk);
  }
}

instantiated under SoC as,

class Soc extends Module {
  Soc(Logic clk) : super(name: 'soc') {
    clk = addInput('clk', clk);

    **my_mod**(clk, name: "**mod_inst**");
    my_mod_copy(clk, name: "mod_inst_copy");
 }
}

RTL generated is:
module Soc(
input logic clk
);
**my_mod_copy**  **mod_inst**(.clk(clk));
my_mod_copy  mod_inst_copy(.clk(clk));
endmodule : Soc

mod_inst is supposed to be the instance name of class my_mod but that's not the case with generated RTL.

### To Reproduce

Use the provided three classes, my_mod, my_mod_copy and soc and generate RTL with them

### Expected behavior

module Soc(
input logic clk
);
**my_mod**  **mod_inst**(.clk(clk));
my_mod_copy  mod_inst_copy(.clk(clk));
endmodule : Soc

### Actual behavior

module Soc(
input logic clk
);
**my_mod_copy**  **mod_inst**(.clk(clk));
my_mod_copy  mod_inst_copy(.clk(clk));
endmodule : Soc

### Additional: Dart SDK info

Dart SDK version: 2.18.3 (stable)

mkorbel1, 2023-07-25:
Perhaps related to #345

mkorbel1, 2024-01-08:
This was fixed in #440

closed 2024-01-08 by mkorbel1