intel/rohd #429: Avoid module creation for simple constant scenarios in gates

mkorbel1, 2023-11-03:
### Motivation

Sometimes it's easy to initialize some variable to 0 or 1, and then do multiple operator functions on it afterwards like `&`, `~`, and `|`.  There are some common rules that we can use to avoid creating gates when not necessary.  For example:
- `~0 == 1` (inversion of a constant is constant of inversion)
- `(x & 0) == 0`
- `(x & 1) == x`
- `(x | 0) == x`
- `(x | 1) == 1`

Skipping over these scenarios would allow for boosts in simulation performance and cleaner generated verilog.

### Desired solution

Modify the operator overrides in `Logic` to avoid creation of `Module`s if it can instead directly compute a `Logic` to return (like in above situations).

Note that this won't completely avoid all the applicable scenarios alone since there could be things like 0 assigned to another signal which is then used in a gate.  Perhaps some optimization in the generation of verilog after assignment collapsing could help even more.

mjayasim9, 2023-11-30:
Will work on this!

mkorbel1, 2023-12-13:
We should probably make it impossible to `put` onto a `Const` to avoid weird surprises when this optimization happens

mkorbel1, 2024-10-14:
I think this should also cover shifts by 0

mkorbel1, 2025-01-08:
This could also cover things like `(x == 1)` since that's equivalent to `x`, or `x == 0` is `~x`