intel/rohd #470: Fixed point support

samimia-swks, 2024-02-29:
### Motivation

Efficient DSP hardware is typically implemented in fixed point arithmetic: 
see https://en.wikipedia.org/wiki/Q_(number_format)
https://ieeexplore.ieee.org/document/7199829

It is painful to write a fixed point data path in SV, as it offers no help in managing the Q point (the number of fractional bits). The designer has to undertake the error prone process of 
- Aligning the binary point of addition / subtraction operands by left shifting 
- Keeping track of how the binary point moves after a multiplication 
 

### Desired solution

Add support for signed and unsigned fixed point 'type's in ROHD, and overload operators such as +, - and * to shift as necessary and keep track of the Q point of the result

mkorbel1, 2024-02-29:
Thank you for the feature suggestion!  I'm wondering whether there's sufficient complexity that it needs to be integrated into ROHD itself or if it could be added to the library at https://github.com/intel/rohd-hcl.

samimia-swks, 2024-02-29:
Hi Max. I am very unfamiliar with Dart and ROHD still, but in my mind a user friendly implementation would look something like this wish list: 
- a fixed point signal that is a subclass of Logic, for example the following could define a Q1.25 signed number: 
`var fp = FixedPoint(name: 'b', iw: 1, qw: 15, signed: true)`
- a fixed point constant that is a subclass of Const with the ability to assign a float (with rounding as necessary) and a toFloat() method that return toInt() / 2^qw
- Support for Arrays of Logic or Const
- Operator overloading for + and - (for arrays as well), such that for example, adding a signed Q2.10 signal/constant to a signed Q5.20 signal/constant is done by shifting the Q2.10 to make a Q2.20, doing the addition, and returning a signed Q6.20.
- Operator overloading for * (for arrays as well), such that for example, multiplying an unsigned Q2.10 signal/constant to a signed Q5.20 signal/constant is done by padding an MSB of 0 to the Q2.10 to make it a signed Q3.20, doing the multiplication, and returning a Q8.40.
- Methods for casting to a different Q format, which would perform dropping LSBs (truncation), padding LSBs (zeros), dropping MSBs (bonus for proper clipping instead of letting it wrap), and padding MSBs (sign extension). 

There is more here but this covers most of the use cases for DSP applications.

mkorbel1, 2024-02-29:
Thanks for the context, very helpful!  I think this sounds quite achievable.

closed 2026-11-03 by desmonddak