intel/rohd #509: "Default" values for `FiniteStateMachine`s

mkorbel1, 2024-09-17:
### Motivation

Often there is a "default" value that would be desirable for most states in a `FiniteStateMachine`, but the current implementation requires repeating the same default across all states in order to avoid an inferred latch.

### Desired solution

Upgrade the `FiniteStateMachine` API to support some way to add defaults.

This could easily be done by inserting conditional assignments (since they act as "blocking") before the rest of the combinational implementation in the `Combinational` that it generates.

awmoore-intel, 2025-02-25:
Is this example below what this issue addresses? In this FSM, I want to initialize len and other signals inside the always_comb block for the default value.  

```
always_ff @(posedge clk) if(reset) len_f <= 64'h0; else len_f <= len;  // flipflop_1

always_comb begin

   /////////////////////////////////////////
   //// Inject Conditionals here
   len = len_f
   /////////////////////////////////////////

  unique case (currentState) 
      2'h0 : begin
          if(cmd_valid) begin
              if((cmd_rs1 != 64'h0)) begin
                  srcAddr = cmd_rs1;
                  dstAddr = cmd_rs2;
              end               else begin
                  len = cmd_rs2;
              end 

          end 

          unique case (1'h1) 
              (cmd_valid & (cmd_rs1 != 64'h0)) : begin
                  nextState = 2'h3;
              end
              default : begin
                  nextState = 2'h0;
              end
          endcase
```

mkorbel1, 2025-02-25:
> Is this example below what this issue addresses?

Yes, exactly!

closed 2025-06-24 by mkorbel1