intel/rohd #527: Consts that `inferWidth` to 0-width generate SystemVerilog with 0-width

mkorbel1, 2024-10-14:
### Describe the bug

For example, when shifting by an integer (e.g. `<< 0`), it can generate code in SV like `<< 0'h0`, which is not valid SystemVerilog.

### Expected behavior

A couple of things:
- When shifting by 0, just don't do anything.  This probably folds into part of https://github.com/intel/rohd/issues/429
- When generating SV of a 0-width constant, make it 1-bit instead.

### Actual behavior

`0'h0` illegal SV

### Additional: Context

Related issues:
https://github.com/intel/rohd/issues/57
https://github.com/intel/rohd-cosim/issues/9
https://github.com/intel/rohd/issues/429
https://github.com/intel/rohd/issues/111

Shubham-Padkonde, 2026-09-21:
This no longer reproduces on `main` at `d22fe80` (ROHD v0.6.11) — both halves of the report now generate legal SystemVerilog.

Shifting by 0:

```dart
addOutput('b', width: 8) <= (a << 0);
```

```systemverilog
module ShiftZero (
input logic [7:0] a,
output logic [7:0] b
);
assign b = a;
endmodule : ShiftZero
```

The shift is folded away entirely, so there is no `0'h0` operand left to emit.

A 0-width constant in a swizzle:

```dart
addOutput('b', width: 8) <= [a.getRange(0, 8), Const(0, width: 0)].swizzle();
```

```systemverilog
assign b = a;  // swizzle
```

I also went looking for any other construct that might carry a 0-width constant through to emission, checking the generated SystemVerilog for a `0'`-prefixed literal in each case:

| construct | result |
|---|---|
| `Const(0, width: 0)` driven directly | clean |
| `Const(0, width: 0).zeroExtend(8)` | clean |
| `a.getRange(3, 3)` (empty slice) | clean |
| `Const(0, width: 0).eq(Const(0, width: 0))` | clean |
| `a << Const(0, width: 0).zeroExtend(1)` | clean |
| `mux(a[0], Const(0, width: 0), Const(0, width: 0))` | clean |

None of them emits a `0'h`/`0'b` literal.

Worth noting the first bullet of the desired solution ("when shifting by 0, just don't do anything") is what the first example shows happening, so that part appears to be covered too.

If there's a construct you know of that still produces the 0-width literal I'm happy to dig further — otherwise this one looks closable.

---

Disclosure: this investigation was performed by Claude Code (Claude Opus 5) working as my agent, at my direction. The outputs quoted above come from real runs in my local environment; I am accountable for this comment.

mkorbel1, 2026-09-29:
Thank you @Shubham-Padkonde !  Yes, this looks resolved and ready to close to me!

closed 2026-09-29 by mkorbel1