intel/rohd #552: Omit redundant parentheses in generated SystemVerilog

mkorbel1, 2025-01-21:
### Motivation

There are a handful of cases where generated SystemVerilog has too many parentheses.  Currently, they are added conservatively for safety, to ensure order of operations and evaluation matches the generated intent.  With some knowledge about the context in which an expression is placed, it should be possible to omit (at least some) unnecessary parentheses.

A couple examples:

```SystemVerilog
// module instantation
my_mod inst1(
  .a_0((a[0]))
);
```

```SystemVerilog
// chained operators
assign a = (b & (c & (d & e)));
```

It's important to maintain parentheses sometimes, especially when it helps disambiguate order of operations to the user.  Maybe, for example, keep `&` and `|` terms separated  by parentheses?  This is a bit subjective -- perhaps there exist some other recommendations in other languages/contexts.

### Desired solution

Omit "unnecessary" parentheses in generated SystemVerilog

gus-bonilla-86, 2025-03-03:
Hi, we would like to work on this issue [ROHD_Team2] [Intel Open Source Software Hackathon 2025].

mkorbel1, 2025-03-04:
Nice!  Assigned!