intel/rohd #569: Improve error message of `Signal changed its value after ...`

awmoore-intel, 2025-02-25:
### Motivation

I am getting this error message but have no idea what signal has the issue. 

```
Signal changed its value after being used within one `Combinational` execution. This can 
lead to a mismatch between simulation and synthesis. You may be able to use 
`Combinational.ssa` to correct your design with minimal refactoring.
```

### Desired solution

Indicate what signal/object has the problem

### Additional details

This was seen when using signals inside a FiniteStateMachine.

mkorbel1, 2025-02-25:
Thanks for raising this.  If I recall, there's a potential performance implication for the "obvious" refactor here to get more context during simulation, but I agree it's important for debug to reveal more information, this is worth fixing.

mkorbel1, 2025-02-25:
Meanwhile, this post can provide some insight potentially to help debug:
https://intel.github.io/rohd-website/blog/combinational-ssa/

mkorbel1, 2025-02-25:
> Nice post!

Thanks!

> I tried to use Combinational.ssa in FiniteStatemachine but it required a Conditional, so didn't work for me.

Yes, I think this would require a new flavor of FSM, maybe something like `State.ssa`, to use that feature within there.

mkorbel1, 2025-02-25:
Filed https://github.com/intel/rohd/issues/571