intel/rohd #570: Waveform shows 'x' when driving signal with 'z'

awmoore-intel, 2025-02-25:
### Describe the bug

RTL Output correctly shows z's being assigned, but waveform shows X's. Seems like waveform only shows z's when a signal not driven. 

```
//  combinational
always_comb begin
  unique case (currentState) 
      2'h0 : begin
          len = len_f;
          dstAddr = dstAddr_f;
          srcAddr = srcAddr_f;
          mem_req_valid = 1'h0;
          mem_req_is_read = 1'h0;
          mem_req_data = 32'bzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzz;
```

### To Reproduce

Drive signal to 'z inside action argument of the FiniteStateMachine class.

e.g.
      io.mem.req.data < Const(LogicValue.z, width: 32),

### Expected behavior

Waveform to show Zs

### Actual behavior

Shows x's

[image]

### Additional: pubspec.yaml

```yaml

```

mkorbel1, 2025-02-25:
I believe this is intended behavior, though your example makes consider some other options.  The idea is that an `always_comb` block represents a "combinational cloud of logic", so even though it is written as an assignment, the result from that block of logic would be X rather than Z floating (undriven).  In this particular case, assignment of a constant Z is sort of a non-synthesizable behavior anyways, so the resulting X can be confusing.  Perhaps the current implementation should stand except for cases where there's an explicit constant Z assignment?

awmoore-intel, 2025-02-25:
Sure, that sounds good. thanks

Shubham-Padkonde, 2026-09-21:
Root cause, verified against `main` at `d22fe80` (ROHD v0.6.11): this isn't the `WaveDumper` — the ROHD simulator itself holds `x`, and the dumper faithfully writes what the signal holds.

A module whose output is driven `z` two ways, dumped together:

```
$var wire 4 s2 direct  $end     // direct <= Const(LogicValue.z, width: 4)
$var wire 4 s3 viaComb $end     // assigned z inside a Combinational

bzzzz s2
bxxxx s3
```

and at the end of the simulation:

```
direct  = 4'bzzzz
viaComb = 4'bxxxx
```

Narrowing it down, only a direct connection keeps `z`:

| how `out` is driven with `4'bzzzz` | resulting simulation value |
|---|---|
| `out <= Const(LogicValue.z, width: 4)` | `4'bzzzz` |
| `Combinational([out < Const(...z...)])` | `4'bxxxx` |
| `Combinational` + `If`/`then` | `4'bxxxx` |
| `Combinational` + `Case` | `4'bxxxx` |
| `mux(sel, z, 5)` | `4'bxxxx` |
| `Sequential` | `4'bxxxx` |

It is per-bit rather than whole-bus: driving `4'b01zx` through an `If` yields `4'b01xx`, so only the `z` bit is rewritten.

The conversion is deliberate and lives in `ConditionalAssign.execute`, [lib/src/modules/conditionals/conditional_assign.dart:62](https://github.com/intel/rohd/blob/main/lib/src/modules/conditionals/conditional_assign.dart#L62):

```dart
if (!currentValue.isValid) {
  // Use bitwise & to turn Z's into X's, but keep valid signals as-is.
  // It's too pessimistic to convert the whole bus to X.
  _receiverOutput.put(currentValue & currentValue);
}
```

`Mux` does the same thing at [lib/src/modules/gates.dart:908](https://github.com/intel/rohd/blob/main/lib/src/modules/gates.dart#L908), via `_d0.value.isValid ? _d0.value : LogicValue.x`.

So the generated SystemVerilog and the ROHD simulator genuinely disagree here: the emitted `always_comb` assigns `32'bzzzz...`, and a SystemVerilog 4-state variable holds that `z`, while ROHD's own simulation of the same block holds `x`.

Since the `&` is intentional and commented, I did not want to guess at the intended semantics. Which direction would you like?

1. Preserve `z` through procedural assignments and `mux`, matching the generated RTL (changes simulation behaviour for anyone relying on today's `x`).
2. Keep the current simulator semantics and document the divergence.

I'm happy to put up a PR for either one — option 1 looks contained to the two sites above plus tests.

---

Disclosure: this investigation was performed by Claude Code (Claude Opus 5) working as my agent, at my direction. The outputs quoted above come from real runs in my local environment; I am accountable for this comment.

closed 2026-11-03 by desmonddak