intel/rohd #572: Waveform signals do not match generated RTL

awmoore-intel, 2025-02-26:
### Describe the bug

Using PairInterface and setting the modify argument causes output waveform to include duplicate signals that are not part of the generated RTL. This is mildly annoying having a bigger list of signals.

 
These are the signals in the generated RTL. 
```
module Crc32 (
input logic clk,
input logic reset,
input logic cmd_valid,
input logic [63:0] cmd_rs1,
input logic [63:0] cmd_rs2,
input logic mem_req_ready,
input logic mem_resp_valid,
input logic [63:0] mem_resp_data,
output logic busy,
output logic mem_req_valid,
output logic mem_req_is_read,
output logic [63:0] mem_req_size_in_bytes,
output logic [63:0] mem_req_addr,
output logic [31:0] mem_req_data,
output logic resp_valid,
output logic [31:0] resp_data
);
logic [63:0] _in0__q;
logic [63:0] _in22__q_add_const_1;
logic [63:0] _in36__q_add_const_1;
logic [63:0] _in3__q;
logic [63:0] _in4__q;
logic [31:0] _original__q;
logic _q_add_const_1_carry;
logic _q_add_const_1_carry_0;
logic [31:0] crc32Val;
logic [31:0] crc32ValNext;
logic [1:0] currentState;
logic [63:0] dstAddr;
logic [63:0] len;
logic [7:0] nLookupIndex;
logic [1:0] nextState;
logic [63:0] srcAddr;
```

Signals in output vcd file
```
$scope module unnamed_module $end
  $var wire 1 s0 clk $end
  $var wire 1 s1 reset $end
  $var wire 1 s2 cmd_valid $end
  $var wire 64 s3 cmd_rs1 $end
  $var wire 64 s4 cmd_rs2 $end
  $var wire 1 s5 mem_req_ready $end
  $var wire 1 s6 mem_resp_valid $end
  $var wire 64 s7 mem_resp_data $end
  $var wire 1 s8 busy $end
  $var wire 1 s9 mem_req_valid $end
  $var wire 1 s10 mem_req_is_read $end
  $var wire 64 s11 mem_req_size_in_bytes $end
  $var wire 64 s12 mem_req_addr $end
  $var wire 32 s13 mem_req_data $end
  $var wire 1 s14 resp_valid $end
  $var wire 32 s15 resp_data $end
  $var wire 1 s16 busy_0 $end
  $var wire 2 s17 currentState $end
  $var wire 64 s18 len $end
  $var wire 1 s19 clk_0 $end
  $var wire 64 s20 srcAddr $end
  $var wire 1 s21 reset_0 $end
  $var wire 64 s22 dstAddr $end
  $var wire 32 s23 crc32ValNext $end
  $var wire 8 s24 nLookupIndex $end
  $var wire 64 s25 data $end
  $var wire 32 s26 crc32Val $end
  $var wire 2 s27 nextState $end
  $var wire 1 s28 valid $end
  $var wire 1 s29 ready $end
  $var wire 1 s30 is_read $end
  $var wire 32 s31 data_0 $end
  $var wire 64 s32 addr $end
  $var wire 64 s33 size_in_bytes $end
  $var wire 1 s34 valid_0 $end
  $var wire 32 s35 data_1 $end
  $var wire 1 s36 valid_1 $end
  $var wire 64 s37 rs1 $end
  $var wire 64 s38 rs2 $end
  $var wire 1 s39 valid_2 $end
  $scope module combinational $end
```

### To Reproduce

Use PairInterface and implement the modify function that changes the name. I'm doing this to prepend the name of the sub interface. 

```
 @override
  PairInterfaceType addSubInterface<PairInterfaceType extends PairInterface>(
    String name,
    PairInterfaceType subInterface, {
    bool reverse = false,
  }) {
    // Fixup name
    subInterface.modify = (n) => '${name}_$n';
    return super.addSubInterface(name, subInterface, reverse: reverse);
  }
```

### Expected behavior

Signals in output .vcd match the generated RTL

### Actual behavior

Additional signals in output .vcd not in generated RTL

### Additional: pubspec.yaml

```yaml

```

mkorbel1, 2025-02-26:
This is true, though the reason is that they are sort of showing different views of the hardware.  The waveform generated by ROHD simulation does not first do a pass to reduce/resolve signal names or combine `Logic`s, rather it is showing any signal that is simulated as part of the ROHD model (with some filtering to reduce noise of unnamed signals).  In contrast, the generated SystemVerilog will sometimes merge signals to reduce `assign` statements or add signals to support LRM/lint compliance.

I don't know if it's realistic or desirable for the generated verilog to have exactly the same signals as are generated in the waves during a ROHD simulation.  More useful is for the names in the waves to match the names of signals from the model, as you would see through the debugger or as constructed and named in your source code.

However, could reasonably be upgraded to consider things like `Naming` or optionally ignore "internal" signals (see #323), so that we could prune down the signals that are probably less valuable or interesting during waveform debug.

awmoore-intel, 2025-02-26:
Ok. thanks for your response. I think we can close this as it seems like a duplicate.

closed 2025-02-26 by awmoore-intel