intel/rohd #573: Allow ints&bools to be used in operators,eq,neq,etc

awmoore-intel, 2025-02-26:
### Motivation

It's annoying to have to wrap all integers with Const and required to provide the width. Can this be part of rohd natively?

will allow syntax like:

```
     io.mem.req.valid < 0,
     io.mem.req.isRead < 0,
```

instead of this
```
     io.mem.req.valid < Const(0, width: 1),
     io.mem.req.isRead < Const(0, width: 1),
```

### Desired solution

```
extension LogicExt on Logic {
  Logic eq(int other) => this.eq(Const(other, width: width));
  Logic neq(int other) => this.neq(Const(other, width: width));
  Conditional operator <(int other) => this < Const(other, width: width);
}
```

awmoore-intel, 2025-02-26:
Related to this, is adding an extension on int to make it simpler..

```
extension ConstInt on int {
  Const C([width = 1]) => Const(this, width: width, fill: false);
  LogicValue L([width = 1]) => LogicValue.ofInt(this, width);
}

   Logic srcAddr_f = flop(
      io.clk,
      srcAddr,
      reset: io.reset,
      resetValue: 0.C(64),
    );
```

awmoore-intel, 2025-02-26:
This seems to already be working. I tried to use `Logic ^ int` in my code and determined I have to use a constant and that went for all constants everywhere.. Closing this.

mkorbel1, 2025-02-26:
your `extension ConstInt on int` is actually a very cool idea!

additionally, there are still some cases where you _can't_ just use an int or bool, but they are on the rarer side.  The most prominent example is for the `<=` assignment operator, which requires a `Logic`, not a `dynamic`.