intel/rohd #579: Capability to disable timestamp in output verilog

awmoore-intel, 2025-03-05:
### Motivation

I want to be able to correlate my input ROHD and output Verilog and track changes. The timestamp clutters up my diff.

### Desired solution

An option on generateSynth to disable timestamp or header entirely.

mkorbel1, 2025-03-05:
The best way to do that with the current APIs is to use the `SynthBuilder` yourself directly.  You can just do `SynthBuilder(myMod, SystemVerilogSynthesizer()).getFileContents()` to get a list of `String`s, one per module.  You can just `.join` them into one big string if you like, or separate them to different files, etc.  Does that meet your needs?

Maybe adding a flag in the `generateSynth` would be good, too.

mkorbel1, 2026-09-29:
Closing since this other existing approach enables the desired result

closed 2026-09-29 by mkorbel1