intel/rohd #580: Inconsistent code generation with Sequential

awmoore-intel, 2025-03-06:
### Describe the bug

This code doesn't generate any always_ff block:

```dart
    Sequential(clk, []);
    Sequential(clk, []);
```

while this code, generates 3 blocks:

```dart
    Sequential(clk, []);
    Sequential(clk, []);
    Sequential(clk, [io.result.valid < 1.C(1)]);
```

```verilog
//  sequential_0
always_ff @(posedge _trigger0__s) begin
end

//  sequential_1
always_ff @(posedge _trigger0__s) begin
end

//  combinational
always_comb begin
  result = 4'hf;
end
```

### Expected behavior

Consistency... either always print verilog for empty conditionals or never.

### Additional: pubspec.yaml

```yaml

```

mkorbel1, 2025-03-06:
Can you share a little more of the surrounding module?  Is the `clk` pin a dangling unconnected signal or is it a port of the containing module?

ROHD "discovers" modules based on connectivity, so if the `clk` pin is not connected to anything, then it will never find them.  If the `clk` pin is dangling, but that `io.result.valid` is a port, then once that additional `Sequential` appears, the `clk` pin now becomes "connected" to the discoverable design, thus the others are included as well.

awmoore-intel, 2025-03-06:
Ah, ok. clk pin is just local to the module and io.result.valid is a real port. I guess this isn't a bug, just unexpected is all. 

```
class validTestIO extends PairInterface {
  validLogic get result => port('result') as validLogic;
  validTestIO() : super(portsFromConsumer: [validLogic("result", 4)]) {}

  validTestIO.clone(validTestIO otherInterface) : this();
}

class Test extends Module {
  late final validTestIO io;
  Test(validTestIO inio) {
    io = validTestIO.clone(inio)
      ..pairConnectIO(this, inio, PairRole.consumer, uniquify: (x) => "io_$x");
    _rtl();
  }
  _rtl() {
    Logic clk = Logic();
    Sequential(clk, []);
    Sequential(clk, []);
    Sequential(clk, [io.result.valid < 1.C(1)]);
  }
}
```

mkorbel1, 2025-03-06:
Ok, yeah that makes sense, then.  I think we'll close this out as not a bug then, unless you have some other concern?

awmoore-intel, 2025-03-06:
Nope, close it. thanks

closed 2025-03-06 by mkorbel1