intel/rohd #594: Generation of `struct`s and `enum`s in output SystemVerilog

mkorbel1, 2025-05-22:
### Motivation

It can be useful to generate SystemVerilog that has more context than just bare signals and arrays.  For example, a struct instead of a bunch of independent signals, or an enum instead of a plain logic for states in an FSM.

### Desired solution

A scalable solution that allows for `enum`s and `struct`s to be included in generated outputs.  This would need to comprehend uniquification to avoid conflicts and legal types for assignments.  It also needs to be controllable so that you can *avoid* generating structs and enums if you want to (e.g. for tool compatibility)

awmoore-intel, 2025-05-22:
+1 This would be very helpful

mkorbel1, 2025-05-29:
Additionally, it would be nice to include this type of information in the generated waveforms.